StephenBlackWasAlreadyTaken/xDrip #184: xDrip on Samsung Galaxy S7

Clauddvon, 2017-04-28:
Hello.

I'm trying to get xdrip working and am having no luck.  Hoping someone can help me figure out what I'm doing wrong.

So I downloaded the xdrip app on my wife's phone, went through steps 1 - 17 in the guide and, since we have already had our sons sensor in (btw, we are parents of a T1 toddler, 3 year old), I had to select April 14th as the "sensor date".

I then left the phone near him for 30 minutes (almost an hour actually) and it's doing nothing.  It's stuck on a message "Please wait, need 2 readings from transmitter first."

Am I supposed to do something manually here???  The guide doesn't mention anything needed here.  What am I missing?

Thanks beforehand

jamorham, 2017-04-28:
Can you use the System Status menu item and let me know which version you are running? Thanks

Clauddvon, 2017-04-29:
Version:  b79a73d-2017.04.25

jamorham, 2017-04-30:
Okay, so you're running xDrip+ what type of collector are you using? Is this for G5? Which guide are you following? Thanks

DayeandClaud, 2017-04-30:
Dexcom G5 Platinum is the receiver.

Trying to get my wife's Android (Samsung Galaxy S7) to host xDrip+ so that I can receive the data (via Nightscout or Share presumably...but I've not gotten that to work at all yet) on my iPhone 7.  So far, I've gotten xDri+ set up on her phone and it "was" receiving data for a while, then it crapped out and I let it sit there with the hopes of reconnecting for about 6 hours but it never did.  Being that my son is 3 year old...and we were upon bed time, I had to turn my CGM back on.  This may stopping the xDrip+ connection since it's now competing with the CGM for the Bluetooth, but I've turned the CGM off repeatidly and restarted the collector, refreshed, turned her phone off repeatedly in the hopes that it would grab the connection again...but it hasn't.

I set up Heroku for Nightscout, but before I continue pursuing getting my phone set up I need to have her Android actually able to get the transmitter reception a large percentage of the time.  I have seen that the Samsung Galaxy S7 has one of the highest connection percentages...it's just not that case for me.

I'm using http://www.nightscout.info/wp-content/uploads/2016/06/xDripG5-Getting_Started.pdf as my guide.

jamorham, 2017-04-30:
Okay that guide is for the xDrip beta which is not the same as xDrip+ there have been a lot of changes since then. Of course there is no actual guide for xDrip+ G5 so I can understand why you'd be following the old guide. Things should be much simpler now with very little needing to be done compared to the beta.

What do you have for your G5 extra options? The only option you should have enabled is `Always Authenticate` and the others like scan constantly etc left unchecked. That is the default configuration and you shouldn't need to change it. The guide looks like it tells you to alter those settings. If you did so then you should change them back.

Set that and check that you have it set to G5 as the data source and that your transmitter number is entered correctly. You should be able to use the system status menu item to see select the G5 tab and get an idea what it is currently doing. Then you should leave the phone within range and see if you start getting data. It might take 20 minutes or so initially.

DayeandClaud, 2017-04-30:
I ended up checking the two options that were recommended to be checked together (or unchecked together).  It wasn't working at all until I checked those two boxes.

amlynek, 2017-08-25:
1. Make sure your xBridge is no more than 3-4 ft away from your son
2. On your receiver phone, turn off WIFI
3. On your receiver phone, turn on BT AND GPS (location)
4. Scan for your xBridge in xDrip (app) NOT your phone's BT settings
5. Always have only one receiver - so either Dexcom OR cell phone - never both. I was never able to use multiple devices to receive readings.
6. Keep both receiver AND xBridge away from other radio devices - wireless controllers (PS/Xbox), BT speakers, microwaves, cell phones with WIFI/BT on, etc.

Keep GPS turned on at all time, it will bond the xBridge with xDrip (give it couple of readings). I have noticed that the syncing/reception between xBridge and receiver phone is better with GPS always on. Battery drain is minimal with it on. Good luck.