StephenBlackWasAlreadyTaken/xDrip #186: xDrip+ Battery Drain

niccoloserci, 2017-05-14:
Hi there, I'm using xDrip+ latest beta version with an Huawei P8 lite and just bought a pre-assembled xDrip Kit to connect to a Dexcom G4 platinum. 

I noticed a massive battery drain first using the kit especially compared to when I used xDrip+ with Librealarm and SW3. After the bluetooth scan and having a look at the system status I found that:

- The connection status is "not connected";
- Bluetooth pairing is "not bonded"
- it seems it's always trying to connect and get random values but for most of the time I get "lost signal"
- Inside the box, the lights aren't always on and have differente status: completely off, flashing, fix.

Please find attached some screenshots. Could you please help me?
Thanks
Samuele

[image]
[image]

amlynek, 2017-06-27:
I will try to assist - same setup - G4/xbridge/xDrip

I noticed a massive battery drain first using the kit especially compared to when I used xDrip+ with Librealarm and SW3.
** I am assuming the drain happens on the phone's battery? If so, there is not much you can do about it. Had a Huawei phone myself and that thing had horrible power retention, battery was being drained by 50% just being in sleep mode during the night. Would never buy a Huawei again for that reason.

If the power drain happens on the xBridge, someone might have installed the wrong type transistors, which allow you to see battery status in your xDrip app, or you have bad LiPo battery.

The connection status is "not connected"
* You need to ensure your GPS/location is on, in order for this to connect.

Bluetooth pairing is "not bonded"
** If I would take guess, you are not "boned" because you are not "connected", per your first screen shot. If I am correct, this is just an inconsistent usage of words between the two menus. Both lines should be called either "connected" or "boned", or better yet, "paired". 

Looking at your first screen shot, even if your are not "connected", your xDrip should still receive data (at least mine does). Once you turn on BT and GPS, scan for your bridge in xDrip. Once detected, pair it. After 10-15 mins you should be receiving data. If you really want that line to show "connected" (or "paired"), click on the refresh circle below "restart collector" couple of times (with GPS still on).

it seems it's always trying to connect and get random values but for most of the time I get "lost signal"
** I'm assuming "signal lost" is shown in xDrip? If so, you need to be closer to your xBridge. Inside the house that would be less than 30 feet, or even less if you have multiple walls between you.

Inside the box, the lights aren't always on and have differente status: completely off, flashing, fix.
** xBridge only flashes when transmitting, two red LEDs, one orange. 
Two red LEDs blinking, orange off, means that xBridge is not connected to a BT device (your phone). All 3 sold LEDs, means they are connecting to a BT device. Solid orange, blinking two red, transmitting a reading. The sequence may need correcting as I am not 100% sure.

Hope this helps.

rlweiner, 2018-07-24:
I've been getting started with the xDrip+ and have a few questions. I'm using it on a Samsung S7 with Android 8 (Oreo). My CGM is a Dexcom G6.

1) Should I be able to run both xDrip and Dexcom’s mobile app simultaneously on the same device? It seems like they conflict, but I haven't found any documentation of this. I've had occasions where (with both running) one was giving me a reading and the other app was saying it had no signal. Also, Dexcom's lock screen badge periodically disappears, and I wonder if xDrip is the cause.

2) Can xDrip show a lock screen badge like the Dexcom app does (the BG reading and trend)? I’ve turned on all the notifications and set it to the large font, but I don’t see anything comparable from xDrip on my S7.  (The only reason I'm still using the Dexcom app is for the prominent lock screen icon)