StephenBlackWasAlreadyTaken/xDrip #188: Extending the range of the  BT Module HM-10

amlynek, 2017-06-27:
This is more of a question, not an issue, so hopefully this is the right place to post it.

Per my title... Is there a way to boost the signal on the bluetooth module HM-10 via a software or hardware mod? I need to get additional +/-10ft out of my xBridge to 'connect' two bedrooms.

With the original G4 Platinum receiver, I was able to get a reading anywhere in the house, even across 3 levels (two physical floors). The interesting thing is that the G4 transmitter is so tiny size, yet the connection was made. This can only mean that the G4 receiver must be doing the hard work and is super sensitive in signal pickup. 

Saying all this brings me to my original issue. With an xBridge setup, which has larger battery and more serious transmitting power, how come that it cannot achieve a connection over the same distance? Is there anything that can be done to remedy the problem? Solution's size factor is not an issue as this will be for night time use only. 

Thanks for reading.

jamorham, 2017-06-27:
I use raspberry pi's around the house to pick up the signal and they connect to wifi and ethernet. You also can use android TV boxes with the dexterity app or nodemcu devices for the same effect.

When out I use a parakeet device which has its own gsm modem. All of these solutions get around bluetooth range problems in different ways.

See https://jamorham.github.io/ for a few more details.

Does that help?

amlynek, 2017-06-27:
Jamorham, first of all I would like to say that you are a genius sir. Thank you for your contributions! You, and all the people behind xDrip, xBridge, NightScout... You are the heros to the rest of us in the fight against diabetes. Giving us options and making the fight more manageable.

With regards to your answer, I get what you are saying. However, your solution requires more know-how, time, and money.... and primarily I don't have the first one. Maybe if there was a "diabetes home automation guide for Dummies" I would be up for it, but as it stands right now, it is rocket science. I was hoping for a simple solution. For example, solder X amount of copper wire to connector A on the BT module - something similar to a two-way radio, boosting the transmission signal. Saying this, I can just leave my receiving phone in a central location in the house, connect it to WiFi, and use NightScout on my other devices to monitor the BG. I just need something for night time highs/lows for one of my kids. The only thing with NightScout is that it is limited in features compared to the xDrip app.

Thanks for your suggestion. Much appreciated.

palaslet, 2017-07-01:
I highly recommend the xDripKit from xdripkit.co.uk
I already use the parakeet for my kid T1D. If my apartment was bigger I would combine it with the wifi module. They both (wifi and parakeet) use the same way to communicate with xDrip, so no extra setup is required, and they seamlessly cover most of your needs. Also, Steve, who creates xdripkit, is really helpfull and quick to answer any questions.

I know it's not free, but it's probably the closest you'll get to a "dummies" system :)

amlynek, 2017-07-10:
A while back I have messaged xdripkit.co.uk but never got a response. 

Going back to my issue, I have realized that next to my receiver-phone, I have my own phone with usually wifi/BT turned on. Turning off wifi/BT on my phone cleared the air noise and allowed the receiver-phone to receive the signal from its designated spot. Now I have the two bedroom connected. Such an obvious answer, yet easily missed. For now, this is my  semi fix.

closed 2017-08-25 by amlynek