StephenBlackWasAlreadyTaken/xDrip #200: Upload data to dexcom share outside US server

JoanoToxo, 2017-11-03:
Would like to send the glucose data to the upload to dexcom servers.
I have libre sensor, blukon and xdrip+ app.
I had installed the dexcom follower app in other android phone, I see the graphic but I don't see the glucose data.
[image]

PatrikTrestik, 2017-11-18:
Hello. Same problem here.
XDrip sucesfully sent follower invite, so I guess connection is established. But follow app still reports status disconnected.

I'm ready to help solving problem. Let me know where to look if more info is required.
Once I have more time I'll try to sniff network traffic.

PatrikTrestik, 2017-11-18:
This is response from Dexcom server. What SN I shoud use?
{
  "Code": "MonitoredReceiverNotAssigned",
  "Message": "Publisher account does not have a valid monitored receiver.",
  "SubCode": "\u003cOnlineException DateThrownLocal\u003d\"2017-11-18 14:47:12.2967619-08:00\" DateThrown\u003d\"2017-11-18 22:47:12.2967619+00:00\" ErrorCode\u003d\"MonitoredReceiverNotAssigned\" Type\u003d\"14\" Category\u003d\"5\" Severity\u003d\"1\" TypeString\u003d\"InvalidState\" CategoryString\u003d\"Logical\" SeverityString\u003d\"Minor\" HostName\u003d\"\" HostIP\u003d\"\" Id\u003d\"{72FB06A3-AD3D-4895-8C0A-39E2E37D4EA8}\" Message\u003d\"Publisher account does not have a valid monitored receiver.\" FullText\u003d\"Dexcom.Common.OnlineException: Publisher account does not have a valid monitored receiver.\" /\u003e",
  "TypeName": "FaultException"
}

apeterson22, 2017-11-18:
Are you sure that you cam send the data from a non-dexcom device to dexcom
servers? Based on the error message posted it's not recognizing the device
as a dexcom device.

PatrikTrestik, 2017-11-19:
I tested two configs. 
1) xDrip+, receiving G5
2) xDrip+, sync follower
Originally I used SN of my old G4 receiver, but than tried also random SN.

At start it was not working in either config. Now it started to send nearly without reason. I guess it was change in SN which helped.

What is correct, to send from follower or reader?

JoanoToxo, 2017-11-20:
I haven't receiving G5. 
I have libre sensor, blukon and xdrip+ app. Is it posible send the data from a non-dexcom device to dexcom servers? I don't know. It can be the problem.