StephenBlackWasAlreadyTaken/xDrip #201: different Insulin parameters during the day

JohnDoeAkira, 2017-11-09:
Hi Emma,
First of all, I want to thank you so much for this app. Wonderful job.
I was looking in the parameters if it was possible to set different insulin sensitivity (and insulin to carb ratio) durint the day. As you know, we have frequently varying parameter (lower sensitivity in the morning for instance). Is this possible with your code ?
Do you think it could be a valuable improvement ?
Cheers,
Akira

jamorham, 2017-11-09:
The original xDrip doesn't support insulin parameters at all. xDrip+ does..
( https://github.com/NightscoutFoundation/xDrip )

With xDrip+ you just long press on the background of the profile editor to split in to as many time blocks as you like. It should explain this in a pop-up the first time you view the profile editor.

JohnDoeAkira, 2017-11-09:
Wonderful ! Thanks for your help !
Sorry not to have read the instructions carefully enough.