StephenBlackWasAlreadyTaken/xDrip #208: "Location Is Not Enabled" even when enabled/granted permissions

jack-margeson, 2018-01-15:
Build - xDrip-plus-20180114-95fcdbc (nightly build 1/14/2018)
Android 7.1.1, MOTO E4
DEXCOM G5

Hello. I would like to use xDrip to connect to my sisters Dexcom G5 device so that she may monitor her numbers from her mobile devices. But, I am having problems with the bluetooth features of the xDrip app. Whenever the app is launched, a prompt appears to ask me to enable the location ("Location Is Not Enabled"). However, the app has full permissions under the app menu of my device - and location is already turned on. This is preventing me from using the bluetooth devices list so that I may connect to her devices. Thank you.

ailuirazal, 2018-09-12:
Hello, 

I'm having the same issue as Jack, the only difference is that my new phone is an MOTO E5. 
below are my specs:
Android 8.0.0 (Oreo),  Moto E5 Plus 4G/LTE
DEXCOM G5
Nightscout version of xDrip+ https://jamorham.github.io/#xdrip-plus
and I keep getting the error "Location is not enabled" even tough it is.

Any suggestions on how to fix it, are welcomed!
Thanks.

ailuirazal, 2018-09-13:
I've solved the error "Location is not enabled" in xDrip+ by going in the phone Settings->Security & Location ->  Location -> Mode where I've selected High Accuracy and not Battery savings. And I stopped getting that error in xDrip + app.

closed 2018-09-26 by jack-margeson

afcady, 2019-02-02:
I still get the error with the location mode "high accuracy."

Otterinaround, 2019-02-21:
I also have this error. (Recently installed Tmobile's android update) I may get 4 readings at best before xdrip requests location permission(which it already has) switching the permission of and back on gets me a few readings but then the error starts up again! Quite frustrating...and also with night time lows,  VERY DANGEROUS.

rgcstats, 2019-04-17:
I get this issue too. My location mode is high accuracy. I'm using a Samsung A5 with Lineage OS.

lkpod, 2019-10-23:
I'm having this issue as well with a Pixel 2 running Android 10.

whynot2018, 2019-11-30:
I get the same issue with my Samsung G. A5 2017. I think we have to close power saver modes.

tkalfaoglu, 2021-05-07:
I don't understand why this program needs high accuracy gps data anyway..
Yes I have now set it to high accuracy but still getting the same scan error: location services needs to be enabled on android 6.0+

jlecount, 2022-06-22:
I had this also although all location services were enabled.  What I did to fix was:
* toggle all location settings (in Location -> Location Services -> Wifi Scanning and Bluetooth Scanning) off and then on again
* toglge Location Permission for Xdrip off then back to "Allow all the time"
* Reboot

Now, i don't know which of these is significant, but i was previously getting "On Android 6.0+ location permission is required.  Implement Runtime Permissions com.policies.rv.androidble2.exceptions.BleScanException: Location Permission missing (code 3)" and now I am not.

I can confirm that all my settings were "right" when i toggled them off and on but something about doing so has resolved this for me also.