StephenBlackWasAlreadyTaken/xDrip #211: Problems with connecting Blucon Nightrider to xDrip

Coldlight42, 2018-01-27:
Hi! I use Samsung Galaxy A5 2017 version with 7.0 Nougat. Last week I received Blucon Nightrider, a transmitter used with Freestyle Libre sensors and I'm having hard time connecting it with XDrip. I use recommended settings, go for the Bluetooth Scan and search for the transmitter, enter PIN number then start a sensor. App requires two scans sent from the transmitter and then nothing really happens. XDrip says there's no signal. When I go to the system status, the app displays the Blucon, although it's not connected. In the BT Device there's "Buggy samsung workaround" active but there's no "Bluetooth pairing: Bonded" . Does anyone know how to fix it?

connormilton, 2018-02-04:
Having same problem :(

jo1st, 2018-03-01:
I'm having the same problem. missed signals. decided to reset xdrip and now blucon does not appear anymore in the scan. the funny thing is that the battery level never changed after over a week and still showed 100%, is that possible using the blucon 24 a day???

heba29, 2018-03-02:
Yes, my last Batterie change was more than 14 days ago.

TheDiabeticWay, 2018-05-21:
How long do the batterys last? Im having same issues  what do you mean by reset blucon? do you mean connect it to that then back to the xdrip?

heba29, 2018-05-22:
The quality of the CR2032 batteries is of course important for the battery life. Cheap batteries last a maximum of 10 days, good 14-16 days. My experience. When coupling with the Nightrider I reconnect directly via the reset switch.

vostok4, 2018-06-11:
My uncle is going through problems now when the battery dies. It looks like the following occurs:

1) Battery displayed at 100% until it is dead
2) Upon inserting a new battery, Nightrider won't reconnect to phone
3) Uninstalled xDrip / Reinstalled xDrip solved the issue

petedcurtis, 2018-07-01:
Hi, I was having problems with NightRider Blucon BT connecting to xDrip.  Very intermittent. Started good then intermittent, then stopped. Indicated blucon battery was 100% in xDrip app.  Used with my Blu X2 phone Android 6. Thought it was my phone, so used my old Blu Studio 7 phone running Android 5, but was the same thing. Finally popped out and checked the CR2032 battery voltage and it was at 2.9V. Put in a new Sony 2032 (3.29V). Then the xDrip app initially couldn't find the Blucon on a BT scan. So I pressed the Blue con reset with the pin and it appeared in xDrip BT scan, put in the code and it connected on and started giving values right away.  So looks like the Blucon NightRider battery indication is not working as it reads 100% all the time in xDrip even with a low battery. Not sure if that's a Nightrider SW issue or if the xDrip app somehow displays the same value.

mysticprog, 2018-07-04:
Hi,
I also have problem with Bluecon and XDrip on Samsung S9. This was my first usage and everything works fine. After around 15 days stops. Now I change the battery, I try few manufactures and decided to use Duracell. I reinstall software, restarting many times mobile phone, few times reset Blucon, nothing happen. Status is "No data received yet" and BT connects but status is not connected. When I try on other mobile phone, everything works. 

Unfortunately, I need it on S9 because this is my son mobile and he have Diab T1.

Any suggestion ?

lkounadis, 2018-10-17:
Hello everyone, has anyone had problems with the connection of the nightrider? Mine keeps on constantly disconnecting and reconnecting by itself. This isnt happening due to distance because the phone is sat within 1 meter of the nightrider. It is really annoying because I cant set notifications now because I keep getting the disconnected notification. This may happen several times within an hour

davidsmooth, 2019-08-07:
Hi everyone, I have been using Blucon for last 8 months without any problem. Ambrosia keeps updating the app with new features. Initially i also had bad experience because I was using it with third party app but since I am using Linkblucon App, it's working fine. Also, another reason they told me for the disconnection is low quality battery brand whereas they highly recommend to use SONY CR2032. Consequently, my Nightrider is working fine. I recommended to few of my friends too.

f0ff886f, 2019-08-07:
I think many folks have connection issues with xDrip+.

I emailed Ambrosia since they announced a new Nightrider, and they said:
>Yes, we agree, our old few devices had some issues and it has been resolved now.

Not super informative, but hopefully the new version is better. I will order it and update here how it goes.

lkounadis, 2019-08-07:
@davidsmooth is the connection with the blucon app stable or does it drop and reconnect? I have been facing constant drop and reconnection issues ever since we got it. also some times even though the phone is right next to the nightrider it may disconnect and never reconnect until we manually reconnect it from the app

davidsmooth, 2019-08-13:
@lkounadis, It's better than xDrip. Which app are you using? They have two apps one is linkblucon for diabetics and second is followblucon for caregivers. Very first day when i downloaded it i faced some issues to understand it but they have good customer service and explained well. Now  I am quite happy with it.

davidsmooth, 2019-09-19:
Hi Mine is working absolutely fine. No connection drops. Please write to
Ambrosia if you are still facing any issues.

Thanks! Regards

David Smooth