StephenBlackWasAlreadyTaken/xDrip #216: (G5 Transmitter) Works only for half an hour then stops for hours.

Rocketman173, 2018-04-03:
I am using the Dexcom G5 transmitter, and it seems to work for a while (30mins-2hrs), and then stops receiving data entirely for a much longer time (30min-10hrs).

It doesn't matter what time it is, whether I have an internet connection, what debug settings are on, etc. It really just stops working for no reason. Any help would be appreciated.

Device: ZTE982

System Status:
Version: d8e7097-2018.01.22
Code: 1801222248
Downgradable to: 1603091400
Data Source: DexcomG5
Bluetooth Device: None set
Connection Status: Not connected

miletine, 2018-04-04:
I also struggled for a while to get it to work reliably on my blackberry priv but I eventually found reasonable settings.  There are several forums discussing the work arounds but generally it seems to come down to how your phone handles Bluetooth connections and making absolutely sure that only one device and one software package is paired with the G5 at any time.   

The old beta group had very useful information for testing the various G5 settings in xdrip:
http://www.nightscout.info/wp-content/uploads/2016/06/xDripG5-Getting_Started.pdf

If you can't get your phone to work then you can brute force it the long way around but compiling a old version of the dexcom app (with android block disabled) and set up your nightscout site with "share" to get the results back to xdrip.  
https://www.reddit.com/r/diabetes/comments/6xvwks/oc_build_your_own_dexcom_g5_android_app_no_phone/

Rocketman173, 2018-04-05:
I've just had it work for 12 straight hours.

But it just stopped again, and hasn't worked in over an hour.

System stays says it's been scanning the whole time.

Can I fix this?