StephenBlackWasAlreadyTaken/xDrip #217: Microwear H5

antrianami, 2018-04-25:
Hello. I have this smartwatch for my daughter with blucon. My problem is that the xDrip loses the connection with the blucon very often and i have to restart the smartwatch and do bluetooth scan and connect again. It can't be problem the distanse between them so i have to search for something else. Could you please tell me what can i check or what am i doing wrong? Thank you

philippebourget, 2018-04-26:
What smartwatch are we talking here? The cause could be that another device is trying to connect to the smartwatch and *Overrides* the connection, therefore interrupting the connection between the smartwatch and blucon.

shafiro, 2018-07-24:
Hi, I am having the exact same issue. Did you work this out?