StephenBlackWasAlreadyTaken/xDrip #226: xDrip and Dexcom G6

rlweiner, 2018-07-24:
I've been getting started with the xDrip+ and have a few questions. I'm using it on a Samsung S7 with Android 8 (Oreo). My CGM is a Dexcom G6.

Should I be able to run both xDrip and Dexcom’s mobile app simultaneously on the same device? It seems like they conflict, but I haven't found any documentation of this. I've had occasions where (with both running) one was giving me a reading and the other app was saying it had no signal. Also, Dexcom's lock screen badge periodically disappears, and I wonder if xDrip is the cause.

Can xDrip show a lock screen badge like the Dexcom app does (the BG reading and trend)? I’ve turned on all the notifications and set it to the large font, but I don’t see anything comparable from xDrip on my S7. (The only reason I'm still using the Dexcom app is for the prominent lock screen icon)

bjpon, 2018-09-18:
I just tried xDrip for first time, and have similar experience running it simultaneously with the Dexcom G6 app--it mostly seems to work, but xDrip will drop signal sometimes (haven't noticed Dexcom app dropping signal). This is on a Pixel 2 running Android 9. 

And agree that while xDrip has so much more customization potential, its lock screen graphic is much less usable compared to Dexcom.

bjpon, 2018-09-21:
Yeah not so much--I continued to get more and more signal drop, so just uninstalled Dexcom.

rlweiner, 2018-09-21:
I gave up on trying to run both -- one or the other would always fail. 

On the other hand, I have been able to run xDrip+ on 2 phones plus the Dexcom receiver, which I'd been told (in the xDrip Facebook group) wouldn't work. It may be working because the old phone is in a room that's out of BT range from the Dexcom receiver and my "real" phone.

eurenda, 2018-09-24:
I've had success with running G5 Dexcom Receiver with my Motorola G5 Plus
(I know - they have the same name) without interference.  Once I've moved
to the G6, I've tried BLE on several phones along with G6 Receiver (another
BLE Android device form what I can tell) and they run over each other (a
reboot sometimes fixes temporarily on either disconnected device).

There must be a way to run my wonderful XDrip+ along with Dexcom G6
Receiver (or even Dexcom phone app on another phone).  Dexcom officially
supports this on a known set of phones + their BLE receiver - so there
should be a way to figure it out.  Sorry I'm not able to provide more help
on this aspect, other than to say Dexcom does multi-BLE receivers, so our
XDrip+ should be able to do it as well just need to figure out how.

machiyag, 2018-10-05:
I just joined to try to find some answers for what is either a related, similar, or the same/identical problem. The CGM in the Cloud Facebook group couldn't solve it, so I'm trying the developer community. I ran xDrip+ happily on my Pixel under Android 8 for months, alongside the old Dexcom G5 receiver. Then, two things happened very close together, and I don't remember in which order: I changed carriers from T-Mobile to Project Fi, and the phone upgraded to Android 9. The two stopped coexisting peacefully, and I decided to try the Dexcom mobile app so I might continue to feed data to my doctor. I did not want to travel internationally with my oversized, elderly laptop, which I had been using to upload to Clarity. I'm now just using the Pixel, and I miss xDrip+. What do I have to do to get xDrip+ to again run on my Pixel?

jamorham, 2018-10-05:
@machiyag How long ago was this? As far as I know the most recent versions of xDrip+ work fine with the Pixel on Android 9. Which version have you tried?

bjpon, 2018-10-05:
Yeah I'm running xDrip+ (Sept 15th build) on my Pixel 2 with Android 9. 

No issues except when I use my bluetooth headphones my Garmin watch drops the connection and I have to re-connect, but I imagine that's something else.

machiyag, 2018-10-05:
I have no idea, at this point, because I uninstalled it when I installed
the Dexcom app. That was mid to late August. I'll install the current
version (is there a release/version #?) and see what happens. Do you think
I need to uninstall the Dexcom app, or is forcing a stop probably good
enough? I've gotten addicted to seeing the numbers.

Marsha

machiyag, 2018-10-06:
OK, it's all coming back to me, because it's failing in the same way. My phone cannot pair with the transmitter. It paired with the same transmitter when I was running the Dexcom app. Running xDrip, the Pixel can sometimes find what it's now identifying as DexcomMP, but fails to pair, and soon loses it as an available device. What setting is wrong? I'm thinking it must be a phone Bluetooth setting problem, but I never changed anything. What's different in Android 9 that has left me disabled here?

machiyag, 2018-10-06:
I just found the "share config via QR code" option, and have a QR code. As far as I can tell, I can't paste it here.

machiyag, 2018-10-06:
Now set to search constantly for G5, so at least I don't have to tell it to try again.

kateaps21, 2021-06-22:
@machiyag Hello! May be you help me? Now I most likely have your Dexcom G6 transmitter, the master sawed down the number. But it is identified in the same way as you describe DexcomMP, maybe you remember the transmitter number

WillyWuttel, 2021-06-22:
?