StephenBlackWasAlreadyTaken/xDrip #227: Correct way to connect xDrip+ to an active G6 sensor

daniel2887, 2018-07-26:
Hi,

I have this scenario:

1. Started using G6 a couple of weeks ago. I started the sensor through the Dexcom receiver (the actual HW receiver, not app), and after warmup I entered the transmitter ID into xDrip. xDrip soon connected and everything was great for 10 days.

2. The sensor expired yesterday, and I started a new one in exactly the same way. Started the sensor from the Dexcom receiver, and didn't change anything in xDrip since it's still the same transmitter.

xDrip resumed reading from the sensor, but, I've been getting a lot of missed readings today. May or may not be related, but I just wanted to check whether I need to do anything with xDrip in this case to tell xDrip a new sensor is being used.
In System Status, Sensor Start date is still 7/15. Does this matter?

I'm running on a nightly build from earlier this month, version a3adda3-2018.07.13.

bjpon, 2018-11-05:
I have somewhat similar question, but not related to receiver.

I'm running a G6 straight to a Pixel and then to a Garmin Forerunner (no receiver or Dexcom app). 

This has been running great for months, but then today when I tried to start a new sensor I'm having connection issues; after the 2-hour warm-up I get this error:  

“G6 State isn’t currently known. Next connection will update this”

I've tried going into System Status and doing the Restart Collector and restarting the phone, but that hasn't worked. 

I thought I should try doing Stop Sensor, but the warning on that page tells me not to do so unless I'm actually taking out the sensor. 

Any suggestions on what to do?  

Thanks!

endtransmission, 2018-12-19:
I'm having the same issue, I see there hasn't been any updates to this thread but wanted to bump it