StephenBlackWasAlreadyTaken/xDrip #228: Issue #486 not completely resolved: problems after restarting the phone

chip83, 2018-08-29:
Hi, the [issue #486](https://github.com/NightscoutFoundation/xDrip/issues/486) it has not been completely solved.
I use the last nightly of 08/28/2018 and the miao miao reconnects normally to the phone after going out of range. However, the problem reappears after restarting the phone. In that case I again have the same problem (issue 486) and the only solution is to uninstall xdrip, terminate the bluetooh application and delete data and cache, reinstall xdrip.

All with a Samsung Galaxy S7 edge upgraded to Oreo (last security patch).

Also from this morning I have another problem with the smartwatch android wear.
The watchface does not seem to work anymore and after a few minutes of normal operation it switches to a "basic mode" in which it only shows the black screen with the time.
I've already tried to uninstall and reinstall xdrip from the smartwatch but to no avail.

Has anyone had the same problem?

jo1st, 2019-03-08:
Actually for me same problem but easy fix, no need to reinstall xdrip. Just forget device under system status. Go to apps and enable the system apps to be shown, stop, clear data and cache of all 3 Bluetooth apps. Then just go to bluetooth scan and select miaomiao. always works for me.

chip83, 2019-03-08:
thanks I try next time