StephenBlackWasAlreadyTaken/xDrip #235: Different value between xDrip and LibreLink with FreeStyle Libre and MiaoMiao

Pierre000, 2018-11-01:
Hello,
I am facing problems of value with xDrip. 
It shows a different value than the official app LibreLink.

I have a FreeStyle Libre sensor, connected to a MiaoMiao device to continually read the values, and transfers it to the xDrip+ app. 

But, when I scan my FreeStyle Libre with the official App, I get a different value than xDrip+. The difference can go **up to 40 mg/dl**, what is quite big. 

I tried to calibrate it several times with blood tests, but after a few hours I get the problem again.

Is anyone knows how to setup the app in order to have the closest value as possible to the FreeStyle reader ?

Thanks !

Hyuril, 2018-11-06:
I have also a difference between FSL scanner and blood tests usually 30 mg/dl I just calibrate XDrip+ once and it runs ok during the 15 days. I have symptoms of hypo around 60 or 70 on XDrip+ and FSL scanner says usually I'm 50 or less so it's ok. FSL scanner underestimated my glycemia and XDrip+ well calibrated do the job. Take care I think it's an individual parameter, take care bye ;)

Pierre000, 2018-11-09:
Hi Hyuril, 
Ok thank you for your answer. 
I will continue to calibrate it if I see a t big gab of value. 
It helps me to see the tendencies and less the precise value, what's enough for me. 
Take care bye !