StephenBlackWasAlreadyTaken/xDrip #237: Signal Missed

illumenatii, 2018-11-15:
Hello,

Bumped into an issue with my G6 and xDrip+. I use a Galaxy Note 9 and I keep running into the issues where "Signal Missed" Pops up and I'm waiting in the range of 15-30 minutes (Sometimes more) for the signal to come back. Not really sure what to do here as I am an amateur at this software.  Any help would be appreciated!

MagicDoggie, 2018-11-26:
I am having the same issue! So frustrating! I have been using this software for about a year now and all of the sudden having constant signal missed issues, I'm going at least 30 minutes, overnight it was 8 hours. I ended up changing the sensor to see if the issue was on that end and now seems every time I  leave the house or am off of wifi its not working for me. I also just did a software update on my note 9 last night, I wonder if that has anything to do with it?

shodakon, 2018-11-29:
Same here.  Note 9 and missed signals.  I had to start using the dexcom app.  Any updates on this?  I'd really really like to switch back.

MagicDoggie, 2018-11-30:
Me too! I haven't had any success with it. I'm using dexcom app also but
not happy with it. Just not enough customization. 😖

Hatschi1978, 2019-01-09:
Are any Updates ? losing still signal ... Any ideas ?

MagicDoggie, 2019-01-09:
Mine is working now. If you check your settings try turning off the option
that says "use the ob1 collector"

VF41, 2019-02-26:
Did turning off "use ob1 collector" solve the issue for anyone?

MagicDoggie, 2019-02-27:
YES!! mine has been working flawlessly since!

VF41, 2019-02-27:
Great to hear! Mine is working well so far (4 hours) without disconnecting, but I'm seeing some  missed readings, possibly due to bluetooth connections to headphones and my watch.

Can anyone anyone share which of the following they have enabled under G5/G6 debug settings? (If those even apply to the G6, since Xdrip seems to treat the G5/G6 the same)
"scan for g5 constantly", "force g5 to UI thread", "authenticate g5 before each read", and "unbond G5 before each read"

MagicDoggie, 2019-02-27:
I am also interested.

SarahjStewart, 2019-03-02:
where do I find the 'use ob1 collector' setting please

I have lost signal twice this morning, my watchface went blank. I have uninstalled and re-installed xdrip+ to watch and it is now showing but the signal for both watch and phone is missed for 30 minutes now

MagicDoggie, 2019-03-02:
Under Settings

G5/G6 Debug Settings

Its the top option then

k0rtalian, 2019-03-21:
I am currently having the same issue with missed reading. There it wants to calibrate. I have removed the G6 transmitter from the bluetooth settings And stopped and restarted the transmitter but instead of accepting the 10 days the sensor has been inserted xdrip still goes through the warm up time even after I entered in the calibration code. I think xdrip is confused about if this is a G5 or G6 transmitter.  You might need to isolate the g5 and g6 settings being lumped into one section makes it difficult to know what you should or shouldnt have turned on.

ImmanuelMc, 2019-04-19:
I am having this issue as well. The events log indicates that location services is disabled, but it is not. If by design or accident I am doing something else that prompts the phone to use my location, the connection to the transmitter resumes. I can reliably reconnect by opening Maps until the connection resumes. I have not determined how to prevent disconnection or whether there is a consistent precursor to disconnection. I get a possibly unrelated error BleGattCharaceteristicException, status 133 and BleDisconnectException: with status 19. These errors happen more frequently and do not seem to coincide at all with disconnections or have other noticeable effects.

DavidAG2000, 2019-11-25:
Okay guys, I'm leaving you the ultimate solution. Although I did it in Spanish, it's very understandable. I did a video, and there you have it in that .zip file. Enjoy it!

You are welcome.

[VIDEO - SOLUTION TO MISSED SIGNAL.zip](https://github.com/StephenBlackWasAlreadyTaken/xDrip/files/3886976/VIDEO.-.SOLUTION.TO.MISSED.SIGNAL.zip)

m00selord, 2020-02-29:
I have tried all of the suggestions but I am still having issues does anyone have a resolution that definitely works?

smeisner, 2020-09-07:
FWIW, I just installed the alpha build and the issue went away for me. I am using the Motorola Stylus phone and the Dexcom G6 and was having many, many connections dropped ...and therefore, missed readings. Since installing the alpha build, all missed readings are gone.

Steve

Pentagit, 2021-02-15:
Hi guys, 
I have a OnePlus 7 smartphone with Android 10 and am still on xdrip from October 2020. Why? Because every version after the October/November 2020 release will miss signal from my G6 after some time. Sometimes it takes days, sometimes just hours.
But my solution has been: do not upgrade. I don't know whats wrong with my configuration (been using xdrip over 2 years), I set it up exactly like in that facebook group pics and have absolutely no problems with xdrip from fall last year. But every xdrip version afterwards will loose signal. Something must have been changed in xdrips way of working with G6 sensors after november 2020 I guess. don't know what though.