StephenBlackWasAlreadyTaken/xDrip #238: Pixel3 XL Android 9 will not keep connection

siblues, 2018-12-07:
I can not get the connection to work between the phone and the G5.The phone will get to the calibration part of the set up and will get reading for a couple of hours but when I go back to check on the phone after maybe 90-120 minutes the phone will not get updated readings.The system status says that the connection is there but it doesn't seem to work.I have tried resetting the bluetooth and restarting it and the phone with no luck.I have refreshed the collector in system status and even tried forgetting it with no luck,Any ideas on what is happening or how to fix it.I also have tried disabling the auto connect in the bluetooth settings and I also tried enabling closing GATT with no luck it makes no difference?

Mike9354, 2019-01-09:
My dad is experiencing a similar issue after updating his android Galaxy S8/S9 (can't remember which) the app still works with NFC but no longer syncs through bluetooth

siblues, 2019-03-22:
After the recent Android Q update everything works great but occasionally it disconnects but a simple restart of the collector works fine.