StephenBlackWasAlreadyTaken/xDrip #240: Dexcom 6G authentication errors

meatychunks99, 2019-01-04:
Hi,
I've had a good trawl through other issues but cannot find anyone reporting this same issue...

We've just bought a trial pack of the G6 for our daugher. We've previously been using libre and Miaomiao with Xdrip+ with a Nexus 5X and a SSW3.

I've followed directions to use the source wizard and entered the transmitter ID. I've disabled the watch to simplify things. However, the status shows 'connecting', and 'authorizing', but nothing more. The list of scanning errors grows larger.
Checking the log I can see:
```
Ob1G5StateMachine
authentication notification throwable:(CHECK_AUTH)
com.polidea.rxandoird.exceptions.BleDisconnectedException:Disconnected from EA:
...

Ob1G5StateMachine
Could not read after AuthRequestTX:
com.plidea.rxandroidble.exceptions.BleGattCannotStartException:
GATT exception from MAC address ... with type
BleGattOperation{description='CHARACERISTIC_READ'}

Ob1G5StateMachine
Received Authentication notification bytes:...
```

I'm using xdrip+ version d1d4889-2018.09.08
Code:1809080944

Starting to think I may have invested in an expensive turkey. Any help would be hugely appreciated...

Norax1995, 2019-02-15:
Hey, I have the same problem with xdrip. Have you found a solution in the meantime?

meatychunks99, 2019-02-15:
Hi @Norax1995, you need to go over to the more recent repository - the code here hasn't been updated in ages. I raised the issue over there and it was resolved (https://github.com/NightscoutFoundation/xDrip/issues/732). Not much detail on there, but we got the issue ironed out and it's currently working in the most recent nightly builds.