StephenBlackWasAlreadyTaken/xDrip #241: Can't connect miaomiao

Zaichee, 2019-01-04:
Hello I have android, Samsung phone, with Libre and miao miao attached. It ws working before but now that I changed sensors the miao miao is stuck on "need 2 readings" for the past 2hrs.
I've restarted my phone, restarted the wifi, Bluetooth and rnf connection says that miao miao is connected.
Can anyone help?

nathanherrmann, 2019-01-10:
Zaichee,
Are you using the Libre 14 day system?  If so, it looks like this is currently not compatible with xDrip.  I think your options right now are to wait for the xDrip community to crack the encryption on the Libre 14, or switch to the Libre 10 day system.

naftis, 2019-01-30:
For me this worked:

Less common settings -> Bluetooth settings -> Disable "Trust Auto-Connect"

jo1st, 2019-03-08:
Since when the 14 days libre is not compatible with xdrip??? I have been using it with xdrip for more then a year

nathanherrmann, 2019-03-09:
There are several different versions of the freestyle libre. The 14 day libre in Europe is different from the 14 day libre in the US. I'm not sure about Asia, but since your profile says you're in Tokyo, I would guess you have the same version as Europe, which is compatible. The 14 day libre in the US, has extra "encryption" that currently (to my knowledge) prevents it from working with xdrip.

jo1st, 2019-03-10:
thanks man. my sensors are from Italy