StephenBlackWasAlreadyTaken/xDrip #242: Libre 2

DerHannesder, 2019-01-07:
As they are rolling out Libre 2 versions in Germany now, which will not be working with miaomiao, I would like to ask if there is any hope of using these sensors with xdrip directly soon?

DerHannesder, 2019-01-12:
Seems to be not too urgent for most users I guess? Still trusting in this great community to get the job done some day in the future.

lorentine, 2019-03-01:
This is starting to get urgent. Libre 2 is now coming to more and more markets. UK is already sending out these when you order Libre 1, and now Norway has been approved for refunds of Libre 2 only. Soon, only Libre 2 will be on the market. :(

ConsFBL, 2019-05-20:
Is there any news on making xdrip compatible with libre 2 sensors, miaomiao or bluereader?