StephenBlackWasAlreadyTaken/xDrip #244: How to upload data to Dexcom server (G6) using Xdrip

gobeti, 2019-01-27:
Hi, I think this is the last step for me to be able to run Xdrip in peace.

My daughter´s Endo looks at Clarity reports and needs to have the data there. I am running Xdrip since yesterday and I am not able to see updated records in Dexcom (EU) clarity server. 

I though I could run the Dexcom G6 app in Android + Xdrip but it seems that once Xdrip is up and linked to the transmitter, this will not allow to pair it in the Dexcom G6 app. 

That is fine though if I get the data going directly to Dexcom server from Xdrip.

Xdrip set up:

Unload BG values as Dexcom = ticked
Dexcom US based account = unticked (I am in Spain)
Dexcom account login and password are correct (I know this because I can use the follow Dexcom up without issues). Hence, data is, somehow, being shared with DExcom server.
10 characters receiver =Ok, the same as my dexcom receiver
Mange followers = I can see the same I had when using the official app 

Any suggestion most welcome!  

Thanks

miletine, 2019-01-27:
I had the same problem with G5 plus Xdrip on European clarity servers.   The fix that has worked for over a year was to use the Dexcom app to collect data and send to clarity then set up Xdrip website to pull readings from the clarity server instead of the G5.   My daughters phone has the Dexcom (monitoring the G5 and uploading), Xdrip (in web monitor mode and to log treatments) along with NSClient (It helps but I don't remember why) and an old pebble watch.  The followers use Xdrip to monitor for support.

Unfortunately I don't remember where I got the instructions on linking the Clarity server to the Xdrip site.

gobeti, 2019-01-27:
Thanks. The issue is that I am doing all this to set the Xdrip app in a smartwatch with 4G (FInow 1 Q pro with 4G sim card) so that i) it monitors and uploads via Xdrip to Nightscout and we can follow my little daugther (6 years)

I was hoping not to lose those Dexcom clarity reports since my Endo only looks at that.... you know? Remotely, he can follow my daughter´s readings and provide solutions and so forth and so on... He only looks at them via Clarity since he has a lot of patients and uses mainly Dexcom Clarity for clinics.

Coming back to use G6 official app is fine as long as I do not want to use an Android 6.0 smartwatch to be the collector... but, that is exactly the point = to use the smartwatch as a collector. Unfortunately G6 official app will not run (independently as collector) in a smartwatch with 4G.

Well, I guess we will have to wait a bit until Dexcom gets out a Dexcom watch (4G) as collector.

Best regards

jamorham, 2019-01-27:
This probably doesn't answer the originally asked question but there is now Tidepool support in xDrip+ which provides a publicly accessible cloud based dashboard/history which you can invite people to view.

xannasavin, 2019-03-13:
I have the same issue: I setup xDrip to upload to Dexcom servers with the correct user, PW, and s/n of the receiver but there is no data reaching the Dexcom clarity dashboard. 
I am not able to use the Dexcom App since they don't support Huawei...
Any ideas?
I btw unticked the "coming from USA" checkbox since appearently I am from Germany - but that should not cause the issue, right?

dbline, 2019-03-17:
Same deal here.  Everything entered correctly and no data on the Dexcom clarity servers.  I would think it must be working for some people or we would hear more about it.  Is anyone out there doing this successfully?  Using XDrip alone to monitor blood sugars and also upload that data to the Dexcom clarity mothership???

gobeti, 2019-03-17:
I don’t think it is possible (after some investigation)... security measures in their servers only allow Receiver or official app to upload info in Clarity.

xannasavin, 2019-03-17:
Also found some replies that mention, it's only uploading to the follow server but NOT to clarity. but not a big deal, as long as your doc does not insist on using clarity...

howzers76, 2019-03-31:
What if they insist on using clarity. The whole point of the question is to get both xdrip to work and dexcom servers to update. Dexcom sucks in regards to software. They do make a create sensor and algorithm but their app team CEO must be still stuck on the commodore 64

dbline, 2019-03-31:
Hey now!  The Commodore 64 was a great machine.  I owe my whole career in IT to the C64.  You had to know what you were doing to write code on a 64K OS.

howzers76, 2019-03-31:
Lol yes I remember the 8.5 floppy.  Do the run command and come back 45 minutes later to see if it was executed

howzers76, 2019-03-31:
I still have no luck getting info onto dexcom servers.  My phone is getting glucose from my sensor but their servers are still not seeing anything past Friday once installed xdrip for the first time

gobeti, 2019-03-31:
And so it will be... As far as I know there is no way data goes into Clarity (security server rules) @howzers76

stuartmcgarrity, 2019-04-15:
I believe there is no public API to write to Clarity, so xDrip can't support it. My work around is to use the Dexcom receiver in parallel with xDrip for a month before my endo visit. She then manually uploads it.

FiReBReTHa, 2019-05-30:
Just wanted to say thanks for posting about clarity being a bust. I kept trying with no success thinking it was my login credentials or something I fat fingered. Sad, but oh well

phicks1, 2019-06-25:
If this option is known not to work, why leave it in??

annsera, 2020-02-01:
Is there any change on this?

charli144, 2020-03-16:
> If this option is known not to work, why leave it in??

it works fine for the dexcom follow app

tomerku, 2020-09-24:
Thank you for this post.

I am using Dexcom G5 with xDrip+ on the phone (without the official dexcom app).
My Clinic is using the Dexcom clarity to assist me with my diabetes balancing. 
Does anyone have a workaround for providing the results to dexcom clarity?
Can we import the data to offline?

Thanks,
Tomer

mick711, 2020-12-18:
Wait to find how solve this problem , i read on reddit that some user with internet connexion use DEXCOM app or BYOD setup share in dexcom app and signin to use clarity . Setup xdrip to use Dex Share Follower .
I don't have internet on my phone so i can't use this so wait to find issue

plickplick, 2021-11-14:
This might help someone :)
Have not tried it myselfe just found it through google.

https://docs.google.com/forms/d/e/1FAIpQLScD76G0Y-BlL4tZljaFkjlwuqhT83QlFM5v6ZEfO7gCU98iJQ/viewform?fbzx=2196386787609383750&fbclid=IwAR2aL8Cps1s6W8apUVK-gOqgGpA-McMPJj9Y8emf_P0-_gAsmJs6QwAY-o0

cassiecaye, 2022-01-31:
> This might help someone :) Have not tried it myselfe just found it through google.
> 
> https://docs.google.com/forms/d/e/1FAIpQLScD76G0Y-BlL4tZljaFkjlwuqhT83QlFM5v6ZEfO7gCU98iJQ/viewform?fbzx=2196386787609383750&fbclid=IwAR2aL8Cps1s6W8apUVK-gOqgGpA-McMPJj9Y8emf_P0-_gAsmJs6QwAY-o0

PLICKPLICK!!!!!!! I've been searching for this information for 2 hours! This DID help someone, me!!!!!!! It took 5 minutes to complete the form, 30 seconds for them to email me the link, downloaded and set up in less than 5 minutes! I'm up and running and still have xDrip+ up and running! I don't have to see that disappointed look in my Endo's eyes ever again! Thank you for sharing this!
❤Cassie

plickplick, 2022-01-31:
Thanks for sharing made my evening! 🙂🙂

kleenkanteen, 2023-01-18:
> This might help someone :) Have not tried it myselfe just found it through google.
> 
> https://docs.google.com/forms/d/e/1FAIpQLScD76G0Y-BlL4tZljaFkjlwuqhT83QlFM5v6ZEfO7gCU98iJQ/viewform?fbzx=2196386787609383750&fbclid=IwAR2aL8Cps1s6W8apUVK-gOqgGpA-McMPJj9Y8emf_P0-_gAsmJs6QwAY-o0

In case anyone else was lost, reddit post about it here: https://www.reddit.com/r/dexcom/comments/y4vixk/oc_build_your_own_dexcom_app_update_now_with_g7/

To use xDrip+ with BYOD Dexcom app, in xDrip+, in settings, the hardware source is "Companion App". Should start getting the local broadcast from BYOD Dexcom app then.

PurpleHel, 2024-05-01:
> To use xDrip+ with BYOD Dexcom app, in xDrip+ in settings, the hardware data source is "640G/Eversense". Should start getting the data through local broadcast.

Is this still the case? I tried it just now with the byod app and a g7 and xdrip says signal missed. I would really like a solution where I don't have to see the dexcom app, but dexcom is still sending info to my provider via clarity. xdrip as companion app doesn't work cos it requires the dexcom app notification be showing.

kleenkanteen, 2024-05-01:
No idea if it works with the g7. Try asking the creator on his reddit post
if it still works. Personally I'm sticking to the g6 because I can reuse
sensors, haha.

Also you can always show your endo your xdrip graph, so dexcom clarity
isn't that important. I stopped going to an endo too because ime, they are
pretty clueless :p

Navid200, 2024-06-14:
xDrip and Clarity: https://navid200.github.io/xDrip/docs/Dexcom/Clarity.html