StephenBlackWasAlreadyTaken/xDrip #252: Galaxy S8  Dex G5 Android Update to 9.0 (SM-G950U) - Missing Reading

motnu, 2019-04-03:
On code 1991192241 
I was updated to 9.0 and reading stopped.   
I have read many of the issues and have tried several of the options with no luck.

It seems that if the lock screen is on I miss readings.  
If i leave the UI in the foreground I get readings.  

I have checked "force  UI foreground"  and rebooted with no luck.  

Logs say "Aggressively restarting collector service dues to lac of reception: Backoff: 360 

I have OB1 Collector "checked" - 

Does anyone have "settings" they can share that solve this issue?  

[image]

[image]

[image]

shayes2791, 2019-04-07:
I've been having the same issues.  Very frustrating for sure

Artchu, 2019-06-01:
Hi! Due to the upgrade of phones to Android 9 (Huawei Honor 10 or Huawei Honor 9 (EMUI 9.0 + Android 9)) my xDrip does`t connect Bluetooth to Blucon. The main error – incorrectly entered password, while using LinkDluCon or Gimp connection with Blucon occurs. It is – incorrectly entered password. I can't even make the first Blucon/ xDrip connection. When I try on other mobile phone (Huawei with Android 7/8), everything works.

I performed all operations according to #486, etc..., but reconfiguration of Bluetooth parameters does not exclude an error. Of course reinstalling to the old xDrip app does not help.

Had the problem even after I set a new sensor and new battery in Blucon.

Did you manage to fix the problem?

motnu, 2019-06-01:
I turned off OB

Here are my settings

motnu, 2019-06-01:
Sorry.   I forget I upgraded so the settings are different.

I got it working on my Samsung by turning off the OB collector.

But there was a stable channel upgrade recently so it's probably
different.

motnu, 2019-06-07:
Now using code 1905181006 and is back to working with OB1 collector.

closed 2019-06-07 by motnu