StephenBlackWasAlreadyTaken/xDrip #261: Sony Smartwatch 3 issue

miz012-om, 2019-06-20:
Hello,

I am very happy with xdrip app and bought also sony smartwatch 3 to see glucose. I installed also Wear OS to pair watch... Everything was running. But at the end I had to make factory reset on the phone and from this time I am not able to get back data to SW3...

I have enable function in xdrip in watch section, SW3 are paired. Notifications are coming there, but I still don't have possibility to change watchface to xdrip as well as I don't see xdrip settings inside phone.

I read some forums and found that xdrip is moved to SW3 automaticly and it can take time... hours / day. For me no change after period...

Is there any "hint" how to push it once more from xdrip app>

Is ok, that in case I click on right up settings menu in xdrip where is Reset Wear DB / Send glucose to Watch / Open Watch settings nothing is done?

Chrisalias, 2019-07-07:
The latest versions of Wear OS are broken for older watches. Get a version of Wear OS from 2018 from https://www.apkmirror.com/

markusvanlaak, 2019-07-24:
Which APK should i use? I'm new to the topic and I'm asked by a friend to make his SW3 work with his Dexcom G6.

whoiswes, 2019-07-24:
I think this project is not really maintained anymore.  I personally use xDrip+ which is actively maintained:  https://github.com/NightscoutFoundation/xDrip