StephenBlackWasAlreadyTaken/xDrip #263: Entered blood test values triggers autocalibration in spite of settings

JustasM11, 2019-07-19:
Hello,

When I enter blood tests it is shown as circle not rectangle. I see that autocalibration is triggered in spite of settings. I use finger icon (left one) on window where threatments are entered. App asked one time, do I want to perform autocalibration using popup and I answered no.

I have nightscout heroku app connected and one nightscout follower. The same happens when followers enters blood test values. When I try to repeat and enter

Please see screenshots attached.

I can collect and provide additional info if needed. 
[image]
[image]
[image]

Chrisalias, 2020-07-21:
Select "Automatic Mode (Best)", but leave "Automatic calibration" unticked. A blood test will appear as a square - tap it if you want to use it as a calibration later on.

JustasM11, 2020-07-21:
Thanks, it works!

closed 2020-07-21 by JustasM11