StephenBlackWasAlreadyTaken/xDrip #271: Lemfo Lem 10 smartwatch 4G - Android 7.1.1

gobeti, 2019-09-28:
Hi,

I am testing the above smartwatch. It is a good solution for kids.

Nevertheless, BT seems to have issues (irregular). Sometimes it works very well, sometimes it misses the 5-minutes signal in periods of around 10 or 20 minutes.

I enclosed some pics of the current config. Aggressively restart collection service and watchdog every 4 minutes seems to make it work in the long run.

Nevertheless, the watch is nice, good quality, reasonably stylish... it would be good to have a better capture rate.

Cheers

[image]
[image]
[image]

gobeti, 2019-10-19:
I made it work, eventually. Thanks

closed 2019-10-19 by gobeti

dubidrubi, 2021-03-06:
@gobeti what have you done to make it work? I'm having LEM10 and exactly the same issues - BT loses connection intermittently... Also, they fall asleep even though all permissions are given and power/traffic saving is disabled... But after 10-15 mins all followers stop receiving reading until I touch the watch.
Watch is running the latest 7.1.1 official firmware, rooted and all Lemfo health apps are disabled/deleted... But the same issue occurs even on the clean watch so it is not related to root or missing Lemfo apps definitely

gobeti, 2021-03-07:
> @gobeti they fall asleep even though all permissions are given and power/traffic saving is disabled... But after 10-15 mins all followers stop receiving reading until I touch the watch.
> Watch is running the latest 7.1.1 official firmware, rooted and all Lemfo health apps are disabled/deleted... But the same issue occurs even on the clean watch so it is not related to root or missing Lemfo apps definitely

I just tested all possible combinations until one of them worked out almost perfectly. I am using DEXCOM G6 with it so that may also be an issue if you are using Libre or similar. Go to app permissions and give all permissions to XDRIP. Make sure the app is running 24/7 on the background in settings and, namely, **battery settings**. Mine is not rooted, not needed. Good luck

jason681529, 2021-03-12:
@dubidrubi I just bought the 3/32GB Domiwear DM20 branded version of this watch to use with my Dexcom G6, and was having similar issues as yourself and gobeti. Sometimes it would work for hours, but mostly it would just fail to connect at each 5 minute update, and I’d have to reset the collector manually. For reference, I factory restored my watch when I first set it up, and am running stock 7.1.1 with v.9 firmware, all permissions allowed and xDrip+ deselected from battery saver. XDrip+‘s settings are all stock too.

I was able to fix it by changing one setting only.

First, go into Settings/More/App settings/Special Access/Battery optimization, and just check to make sure that xdrip+ is set to not optimized. In my case, it already was set to not optimized, probably because deselecting it from Battery saver also changed it here.

next, just swipe back and scroll down to Settings/More/App settings/Special Access/Modify system settings, and set Xdrip+ to yes. This is the big important setting, and what gives the app proper control over the dm20/lem10’s Bluetooth.

After changing that setting, everything has just worked for me. I use it with only Bluetooth on and location set to power saver, and get great battery life!

jason681529, 2021-03-12:
@dubidrubi okay, I just had it miss an update. The setting still helped, as it allowed xDrip+ to auto connect on its own 20 minutes late, when the Bluetooth Watchdog took over and reset Bluetooth. So I’d probably change the timer value to 5 minutes, and you’ll be golden.

atj393, 2021-11-16:
@jason681529 Can you help telling how to hard reset the device? Thanks.

gobeti, 2021-11-16:
Easy. Go to settings and reset to factory....