StephenBlackWasAlreadyTaken/xDrip #274: Issues with New "firefly" G6 transmitter (serial# 8G... or 8H...)

cbadams7, 2019-11-11:
Recieved two new transmitters for my dexcom G6 with serial numbers beginning in 8G. Attempted multiple times to follow instructions listed here https://androidaps.readthedocs.io/en/latest/EN/Configuration/xdrip.html but with no success. I will sometimes recieved a few minutes of reading but eventually xdrips+ will no longer report current values. Please help! Really need this to work for an upcoming triathlon!

corticalneuron, 2019-12-24:
I've got a G6 (8G****) with one of the newer transmitters and to make it work, I've had to physically remove the transmitter from the sensor and let it sit for a half hour, then put it back in the sensor and start the sensor again. It's not technically possible to remove it but you can using test strips to carefully pop it out. I was told by a nurse that when you switch to a new sensor after 10 days, you have to let the transmitter sit, not being connected to anything, for at least 15 minutes. All of the other methods of running the sensor without a code or with 0000 as the code have not worked for me, but physically pulling the transmitter out and waiting does. 

Also, if you ever hit "stop sensor" then you can't actually restart your sensor again unless you physically remove the transmitter and let it sit for a while. It is designed to stop you from reusing sensors... They can try but we will continue to find loopholes, I hope. 

Good luck with your triathlon and I hope this is helpful!

Vaughanabe13, 2020-03-09:
I have this same problem with a transmitter that starts with 8G (most recent one I have gotten from Dexcom). I get a few readings and then pretty much it stops receiving readings indefinitely. It seems to be related to the unbonding somehow. The only way I can restart the readings is by going to System Status -> Forget Device. Then it will briefly get readings again (maybe only once?) and then pretty much immediately it stops working again.

The most annoying part of this is that if my BG is out of range, the alert will stay active forever while I don't have readings, so I'm constantly getting BG range alerts at the snooze interval even though the glucose values are stale/invalid.

One thing that helped for a while was disabling the "Trust Auto-Connect" setting under less common settings -> Bluetooth Settings. That seemed to be working on my previous transmitter but now on this new transmitter I still have this problem.

Sometimes when I am sitting very still for a long period of time it will continue to receive readings, but as soon as I start to move it generally goes out.

I have tried various combinations of bluetooth settings/watchdogs/wakes/etc. and I can't figure out the right combination of settings to make it work for a period of time.

Vaughanabe13, 2020-03-09:
Quick update to my previous post. I was running xdrip+ in "Beta" release mode. However I saw a comment somewhere that said I need to be on the nightly builds for the newer G6 stuff to work. I just updated xdrip+ to the 2/20/20 nightly (https://github.com/NightscoutFoundation/xDrip/releases/tag/2020.02.20b).

Now my transmitter seems to be working pretty solidly without having to change additional settings.

My phone is Pixel 3 running Android 10.