StephenBlackWasAlreadyTaken/xDrip #284: xDrip for Huawei without Google (like P40pro)

figlmd, 2021-01-21:
Hello! Thank you very much for developing x-Drip plus! It is really a great app. Is there a chance to modify it to work with Huawei P40pro which doesn't have Google services and Google messeges? Thank you once again!

teamoth, 2021-02-11:
Hi there, I can't comment from the point of view of the developer, but as I've posted an issue myself I thought maybe I can try and help somehow. I see that the Huawei P40pro doesn't come with Google Services (like Android normally does), but can you explain what happens when you try to install/use xDrip+ from GitHub? I wasn't aware that Google Services was in anyway involved in how xDrip+ works on my Pixel (for example). Can you enable 'unknown sources' on the Huawei P40pro?

figlmd, 2021-02-11:
Hello! I can enable unknown sources, but it says, 'no sync by missing Google play services' and xdrip simply won't work nor as a Master, nor as a Follower. It works as a Follower only if I set it up with Nightscout, but then the alarms go crazy-they beepe without a reason - like if there was missed reading for some time, but the last reading is written like 2 minutes ago... Poslano s telefona Huawei-------- Izvorna poruka --------Šalje: teamoth <notifications@github.com>Datum: čet, 11. velj 2021. 16:23Prima: StephenBlackWasAlreadyTaken/xDrip <xDrip@noreply.github.com>Kopija: figlmd <figlmd@gmail.com>, Author <author@noreply.github.com>Predmet: Re: [StephenBlackWasAlreadyTaken/xDrip] xDrip for Huawei without Google (like P40pro)  (#284)
Hi there, I can't comment from the point of view of the developer, but as I've posted an issue myself I thought maybe I can try and help somehow. I see that the Huawei P40pro doesn't come with Google Services (like Android normally does), but can you explain what happens when you try to install/use xDrip+ from GitHub? I wasn't aware that Google Services was in anyway involved in how xDrip+ works on my Pixel (for example). Can you enable 'unknown sources' on the Huawei P40pro?